OAID/Tengine #120: 编译报错skipping incompatible /usr/local/lib/libopencv_cudabgsegm.so when searching for -lopencv_cudabgsegm /usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: cannot find -lopencv_cudabgsegm

ArtyZe, 2019-03-19:
Ｈello, I have met a problem when I try to cross compile Tengine, 
aarch64-linux-gnu-g++ --sysroot=/home/artyze/Desktop/QNX/tengine_aarch/sysroot/ubuntu_rootfs -L/usr/lib/aarch64-linux-gnu -L/lib/aarch64-linux-gnu  /home/artyze/Desktop/QNX/tengine_aarch/build/tests/bin/test_onnx_sqz.o  -o /home/artyze/Desktop/QNX/tengine_aarch/build/tests/bin/test_onnx_sqz  -L ../build/ -ltengine -Wl,-allow-shlib-undefined -lprotobuf -ldl -lpthread -Wl,-rpath,./build/ -L/usr/local/lib -lopencv_cudabgsegm -lopencv_cudaobjdetect -lopencv_cudastereo -lopencv_dnn -lopencv_ml -lopencv_shape -lopencv_stitching -lopencv_cudafeatures2d -lopencv_superres -lopencv_cudacodec -lopencv_videostab -lopencv_cudaoptflow -lopencv_cudalegacy -lopencv_calib3d -lopencv_features2d -lopencv_highgui -lopencv_videoio -lopencv_photo -lopencv_imgcodecs -lopencv_cudawarping -lopencv_cudaimgproc -lopencv_cudafilters -lopencv_video -lopencv_objdetect -lopencv_imgproc -lopencv_flann -lopencv_cudaarithm -lopencv_core -lopencv_cudev 
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: skipping incompatible /usr/local/lib/libprotobuf.so when searching for -lprotobuf
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: skipping incompatible /usr/local/lib/libprotobuf.a when searching for -lprotobuf
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: skipping incompatible /usr/local/lib/libopencv_cudabgsegm.so when searching for -lopencv_cudabgsegm
/usr/lib/gcc-cross/aarch64-linux-gnu/5/../../../../aarch64-linux-gnu/bin/ld: cannot find -lopencv_cudabgsegm

Does that means I need to cross compile Opencv at the same time ? Thanks a lot.

cyberfire, 2019-05-04:
@ArtyZe 

It looks like an ABI compatible issue. Are you trying to build tengine with toolchains for QNX?

Thanks
Haitao

ArtyZe, 2019-05-06:
@cyberfire 
First I want to compile it for linux, later I will try to compile a version for QNX, but I think maybe tengine rely on opencv to load and write image, so first need to cross compile opencv :(

guchaojie, 2019-08-26:
@ArtyZe  how to fix this compile issue finally？

guchaojie, 2019-08-26:
readelf -h libopencv_ximgproc.so
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 03 00 00 00 00 00 00 00 00 
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - GNU
  ABI Version:                       0
  Type:                              DYN (Shared object file)
  Machine:                           AArch64
  Version:                           0x1
  Entry point address:               0x175e0
  Start of program headers:          64 (bytes into file)
  Start of section headers:          1278320 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         7
  Size of section headers:           64 (bytes)
  Number of section headers:         26
  Section header string table index: 25