OAID/Tengine #125: Get a error when run caffe model

pangchaogit, 2019-03-26:
I run caffe model and take a wrong : 
When run tensor_t out_tensor = get_graph_output_tensor(graph, 0, 0),After I found out_tensor is NULL, same time I make sure preceding steps is correct  ,Is my caffe model not request?(the model is moiblenet and resnet)

hey-yahei, 2019-03-30:
Hi, pangchaogit
It seems that Tengine could not detect your output tensors, please try to call `get_graph_tensor` function to get it via specify the tensor name.
```cpp
tensor_t get_graph_tensor(graph_t graph, const char * tensor_name);
```

Thanks

pangchaogit, 2019-03-31:
I try tensor_t out_tensor =get_graph_tensor(graph_t graph, const char * tensor_name) to deal with my problem And out_tensor is not NULL.what a pity ,when I call function float* outdata = ( float* )get_tensor_buffer(out_tensor);the outdata is NULL

lyuchuny3, 2019-04-02:
Hi, pangchaogit,

Please pull the lastest codes of Tengine. The last commit should have fixed this bug.

Thanks

pangchaogit, 2019-04-02:
Thank you ! I solved my problem by your latest tengine_lib.but find tensor_t get_graph_tensor(graph_t graph, const char * tensor_name) is not able and  tensor_t out_tensor = get_graph_output_tensor(graph, 0, 0)  got the correct output

lyuchuny3, 2019-04-03:
Hi, you can get your tensor by:
```
 tensor_t  my_tensor = get_graph_tensor(graph_t graph, const char * tensor_name) 
```
If you don't get the correct tensor, please check you tensor name;
you can add `dump_graph(graph);` after run_graph, and find your wanted tensor name.
Thanks

closed 2019-04-05 by cyberfire