OAID/Tengine #128: Object Detection Program?

Zeit42, 2019-04-06:
I checked the available samples and what I see are Image Classification programs. I need to run an Object Detection on a [Rock960 Board](https://www.96boards.org/product/rock960/ai/). Using the native Android 7.1 OS that it came with, the Object Detection does NOT work on the 2GB RAM Version of the board. 

I saw the repository, and it seems that there aren't any sample Android code that uses Object Detection. 

Any leads on this?

cyberfire, 2019-04-08:
@Zeit42 

Which Object Detection model you've tried? 

Thanks,

Haitao

Zeit42, 2019-04-08:
> @Zeit42
> 
> Which Object Detection model you've tried?
> 
> Thanks,
> 
> Haitao

I just tried running the example that came with TensorFlow Lite and it was shutting the board down.