OAID/Tengine #134: 为什么树莓派3B+在make install安装失败

lanzhegu, 2019-04-14:
serializer/built-in.o -lpthread -lprotobuf -ldl -lopenblas
/usr/bin/ld: 找不到 -lopenblas
collect2: error: ld returned 1 exit status
Makefile:116: recipe for target '/home/pi/Tengine/build/libtengine.so' failed
make: *** [/home/pi/Tengine/build/libtengine.so] Error 1

lanzhegu, 2019-04-14:
No package 'opencv' found
cp: 无法获取'/home/pi/Tengine/build/tools/bin/test_accuracy' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/pi/Tengine/build/tools/bin/test_yolov2' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/pi/Tengine/build/tools/bin/convert_model_to_tm' 的文件状态(stat): 没有那个文件或目录
Makefile:50: recipe for target 'install' failed
make[1]: *** [install] Error 1
make[1]: Leaving directory '/home/pi/Tengine/tools'
Makefile:145: recipe for target 'tools' failed

zhangscth, 2019-04-19:
我也遇到同样的问题了，找不到 -lopenblas。。。

hey-yahei, 2019-04-19:
Hi, @lanzhegu & @zhangscth 
Please install dependent libraries before installation. And if you want to install Tengine on Raspberry Pi, this [article](https://hey-yahei.cn/2018/10/13/RasPi-Tengine/) in Chinese may be helpful.
Thanks

closed 2019-07-17 by cyberfire