OAID/Tengine #136: failed to set node ops for node: ConvNdBackward202 op: Deconvolution

dzyjjpy, 2019-04-23:
errors when **run caffemodel with tengine in rk3399. my model is converted from pytorch, backbone resnet-50, with deconvolutional ops inside.**

 $./HeadPose headPose_new.prototxt headPose_new.caffemodel headpose_1.jpg

when I excute the command, it shows:
WARNING: linker: /data/local/tmp/HeadPose: unsupported flags DT_FLAGS_1=0x8000001
ACL Graph Initialized
Driver: ACLGraph probed 1 devices
create graph success
**failed to set node ops for node: ConvNdBackward202 op: Deconvolution**
/home/jiapy/workspace/Tengine-master/executor/lib/graph_task.cpp:229 failed to Prerun task on dev executor: generic_probe
/home/jiapy/workspace/Tengine-master/executor/lib/graph_task.cpp:315 bad status: CREATED

repeat 1 times, avg time per run is 0.098 ms
out_tensor point null

It will be appreciated if you can provide any useful advice. Thanks

hey-yahei, 2019-04-24:
Hi, @dzyjjpy 
The executor for **Deconvolution** operator is only implemented with OpenBLAS ([deconv_2d_blas.cpp](https://github.com/OAID/Tengine/blob/master/executor/operator/common/blas/deconv_2d_blas.cpp)). Please uncomment `# CONFIG_ARCH_BLAS=y` in `makefile.config` to enable OpenBLAS, rebuild Tengine and try again.
Thanks

dzyjjpy, 2019-04-24:
Hi @hey-yahei . Thanks for your quick response.  I uncomment # CONFIG_ARCH_BLAS=y in makefile.config, point the right path for BLAS, rebuild Tengine successfully.  rebuild the examples, to get my own project's file HeadPose.
However,  the error still exist,  it seems doesn't work 

**$ ./HeadPose headPose_new.prototxt headPose_new.caffemodel headpose_1.jpg**      
WARNING: linker: /data/local/tmp/HeadPose: unsupported flags DT_FLAGS_1=0x8000001
ACL Graph Initialized
Driver: ACLGraph probed 1 devices
create graph success 
failed to set node ops for node: ConvNdBackward202 op: Deconvolution
/home/jiapy/workspace/Tengine-master/executor/lib/graph_task.cpp:229 failed to Prerun task on  dev executor: generic_probe
/home/jiapy/workspace/Tengine-master/executor/lib/graph_task.cpp:315 bad status: CREATED
....
--------------------------------------
repeat 10 times, avg time per run is 0.0178 ms
 out_tensor point null

dzyjjpy, 2019-04-28:
solved by proposal above. closed

closed 2019-04-28 by dzyjjpy