OAID/Tengine #137: About muti thread

maohuui, 2019-04-25:
I am using RK3399, dual core A72 + quad core A53, I want to use different cpu to handle different threads. But I don't understand the example in multi-thread. After std::thread link cpu, use create_graph, set_graph_device, I don't need graphics processing, how to set the CPU to handle multi-thread problems,

hey-yahei, 2019-04-28:
Hi, @maohuui 
For Tengine, you can use `set_cpu_list()` to bind cpu globally, but note that you should call it before `init_tengine()`. And if you want to bind some thread to specific cpu cores, try to utilize **cpu affinity** -- call `sched_setaffinity()` to bind thread to specific cpu cores at the begining of the thread.
Thanks

maohuui, 2019-04-28:
I see ,thank you , hey-yahei .

closed 2019-05-07 by lyuchuny3