OAID/Tengine #139: Run FASTER_RCNN Segmentation fault

guoqiang0148666, 2019-04-26:
Thanks OAID teams grateful works

my platfoms is rk3399 and os is ubuntu 16.04

l run the "./examples/build/faster_rcnn/FASTER_RCNN" , using your provide "VGG16_faster_rcnn.prototxt" and "VGG16_faster_rcnn_final.caffemodel", everything is ok.
However, when l use my custom file (mynet_main_new.pt, mynet_main.caffemodel), l meet the follwoing error:

./examples/build/faster_rcnn/FASTER_RCNN -p /root/Caffe-HR T/mynet_main_new.pt -m /root/Caffe-HRT/mynet_main.caffemodel
/media/linaro/f9a0c086-b406-41d8-a375-ad1fb929f773/Tengine/examples/build/faster_rcnn/FASTER_RCNN
image file not specified,using /media/linaro/f9a0c086-b406-41d8-a375-ad1fb929f773/Tengine/tests/images/ssd_dog.jpg by default
Segmentation fault

l debug the codes and find the above error occur in
 ret = prerun_graph(graph);
that is to say, the function fail to finished.

l also change the caffemodel file and run the following command:
./examples/build/faster_rcnn/FASTER_RCNN -p /root/Caffe-HRT/mynet_main_new.pt (mynet_main_new.pt is upgrate proto file)
and get following log:

/media/linaro/f9a0c086-b406-41d8-a375-ad1fb929f773/Tengine/examples/build/faster_rcnn/FASTER_RCNN
model file not specified,using /media/linaro/f9a0c086-b406-41d8-a375-ad1fb929f773/Tengine/models/VGG16_faster_rcnn_final.caffemodel by default
image file not specified,using /media/linaro/f9a0c086-b406-41d8-a375-ad1fb929f773/Tengine/tests/images/ssd_dog.jpg by default
[libprotobuf WARNING google/protobuf/io/coded_stream.cc:604] Reading dangerously large protocol message.  If the message turns out to be larger than 1073741824 bytes, parsing will be halted for security reasons.  To increase the limit (or to disable these warnings), see CodedInputStream::SetTotalBytesLimit() in google/protobuf/io/coded_stream.h.
[libprotobuf WARNING google/protobuf/io/coded_stream.cc:81] The total number of bytes read was 548317115
height width scale225,300,0.390625

infer shaped for node: conv1 op: Convolution failed
Prerun graph failed, errno: 0

the log means the error is caused by my caffemodel file (mynet_main.caffemodel), l really don't know how to solve the problem, anyone can help me and any suggestion are welcome! thank you!

dzyjjpy, 2019-04-28:
have  you solved the issue? I met similiar error: segmentation error when run the line:
rungraph(graph, 1)
and prerun function can return success. looking forward to your experience.

goldlee, 2020-06-20:
> have you solved the issue? I met similiar error: segmentation error when run the line:
> rungraph(graph, 1)
> and prerun function can return success. looking forward to your experience.

guys, have you solved this problem, i met this problem also, can you share with me, if you solved,thank you!