OAID/Tengine #141: Tengine /examples/mtcnn

MaxwellAlen, 2019-05-08:
I run the default mtcnn by ./MTCNN,it passed,but i run
 ./MTCNN  my_image.jpg  Tengine/models/det1.caffemodel   save_result.jpg ,the terminal displayed 
/home/liu/NN/Tengine/examples/mtcnn/build/MTCNN
Input file not existed: Tengine/models/det1.caffemodel/det1.prototxt
Segmentation fault (core dumped)
please hlep me.

liqi-c, 2019-05-23:
maybe it's Tengine/models/det1.caffemodel/det1.prototxt file not exist ? you can check this ,thinks

bzhang5, 2020-04-16:
@MaxwellAlen You can use gdb --args to debug, and then use bt to check which line has issue, if the info display there is no such file or loadmodel function failed, there might be problem for no model file.

closed 2020-04-16 by bzhang5