OAID/Tengine #142: examples/YuFaceDetectNet run error "failed to Prerun task on  dev executor: arm.probed"

zhuqinglisheng, 2019-05-14:
My device is RK3399Pro, I install Tengine by the steps of /Tengine/doc/install.md, and I don't modify the makefile.config. I want to run the example of "/Tengine/examples/YuFaceDetectNet", I download caffemode and prototxt files by using url(https://pan.baidu.com/s/1LXZ8vOdyOo50IXS0CUPp8g, Tengine_models/yufacedetec), but cmake and make, I run the executable file with cmd "./YuFaceDetect yufacedetectnet-open-v1_new.prototxt yufacedetectnet-open-v1_new.caffemodel demo.jpg", error display as follows:

[root@localhost YuFaceDetectNet]# ./YuFaceDetect yufacedetectnet-open-v1_new.prototxt yufacedetectnet-open-v1_new.caffemodel demo.jpg 
graph_task.cpp:234 failed to Prerun task on  dev executor: arm.probed
type is not same-----------------------
terminate called after throwing an instance of 'TEngine::bad_any_cast'
  what():  Bad any cast:  Expected: float*   Real: void
Aborted (core dumped)

Does anybody would help me, thank you very much.

cyberfire, 2019-06-03:
@zhuqinglisheng 

Does other network, such as mobilenet, work on your board or not?

Thanks,

Haitao

Hippop, 2019-06-18:
Have you solved your problem? I met a same error.

rahulsharma11, 2019-11-05:
Hi, anyone solved this issue?
I met the same (dev executer)error with ."tf_lite_mssd_quant" test code.

chenti2x, 2019-12-17:
Hi, anyone solved this issue? 
I met the  same problem，error is：

type is not same-----------------------
terminate called after throwing an instance of 'TEngine::bad_any_cast'
  what():  Bad any cast:  Expected: TEngine::StridedSliceParam   Real: void