OAID/Tengine #143: no available Tengine fp16 model file (*_fp16.tmfile)

fanxingzju, 2019-05-15:
"examples/tengine_model/classification/classification_fp16.cpp" can't work properly without fp16 model file.

https://github.com/OAID/Tengine/blob/3a0ae197ff6ef110db10ef04b2454d19db6b93a7/examples/tengine_model/classification/classification_fp16.cpp#L64-L74

Neither can I find such model file in Tengine model zoo (psw: 57vb), nor can I convert common cafffe model file(float) to such type by means of Tengine convert tool.

liqi-c, 2019-05-23:
@fanxingzju yes ,you need to convert vgg16.caffemodel -->vgg16.tmfile-->vgg16_fp16.tmfile . 
i store vgg16_16.tmfile in  
链接：https://pan.baidu.com/s/1vaVFGIID2Mdxy2ksGV4hyQ 
提取码：ett5 
maybe you can try .thinks

fanxingzju, 2019-05-24:
> @fanxingzju yes ,you need to convert vgg16.caffemodel -->vgg16.tmfile-->vgg16_fp16.tmfile .
> i store vgg16_16.tmfile in
> 链接：https://pan.baidu.com/s/1vaVFGIID2Mdxy2ksGV4hyQ
> 提取码：ett5
> maybe you can try .thinks

Thanks a lot. 

I try to run tm_classify on the Taishan2280 （cpu hi1616，32*A72）,it seems that this server does not support fp16.

```
root@ubuntu:/mnt/Tengine-projects/float32/Tengine# ./examples/build/tengine_model/classification/tm_classify_fp16 -n vgg16 -t ../../tmodels/vgg16_16.tmfile -l ./models/synset_words.txt -i ./tests/images/bike.jpg -g 224,224
/mnt/Tengine-projects/float32/Tengine/examples/build/tengine_model/classification/tm_classify_fp16
graph_task.cpp:234 failed to Prerun task on  dev executor: arm.probed
Prerun graph failed 
```

Do you have any int8 Tengine model file?

I find that a quantization tool has been mentioned, but the corresponding code is not provided.
 
https://github.com/OAID/Tengine/wiki/Tengine%E4%BD%BF%E7%94%A8%E6%95%99%E7%A8%8B#tengine%E9%87%8F%E5%8C%96%E5%B7%A5%E5%85%B7

Thanks again.

liqi-c, 2019-05-25:
Mabey you need to wait for the quantization tool  publish in github .... 
@fanxingzju 
thinks .

fanxingzju, 2019-05-25:
@liqi-c 
Thank you all the same .  :smile:

closed 2019-05-25 by fanxingzju