OAID/Tengine #144: Cannot find load function for operator:StridedSlice

VXallset, 2019-05-20:
Hello, I converted my Tensorflow model to a .pb file and want to use it on Firefly RK3399. However,  Tengine loaded model failed and indicated " Cannot find load function for operator: StridedSlice". So do I have to implement the StridedSlice myself, or if there are other solutions?

cyberfire, 2019-05-21:
@VXallset 
We just pushed the code to support stride slice. please have a try.

Thanks,

Haitao

VXallset, 2019-05-23:
@cyberfire 
Hi Haitao,

It seems not working. When doing   `TF_Session* session = LoadGraph(model_file.c_str(), graph);`, I got the following information:

> type is not same-----------------------
> terminate called after throwing an instance of 'TEngine::bad_any_cast'
> what():  Bad any cast:  Expected: TEngine::StridedSliceParam   Real: void

and the code aborted. If necessary, I can send you a copy of my .pb file and .prototxt file via email. 

Cheers,

Yu

ZPL95, 2019-05-26:
@VXallset 
Hi,Yu:

Please send me your model files for further support.

BR,
ZPLuo

chenti2x, 2019-12-17:
I meet the same problem too,Is this problem solved？