OAID/Tengine #153: The environment variable TENGINE_CPU_LIST doesn't work!

Guanmoyu, 2019-06-10:
I have tried TEngine in a device with 4 CPUs (A53) and the example YuFaceDetectNet runs well. In order to test the model with deconverlution layer, I set CONFIG_ARCH_BLAS=ON. However when I set TENGINE_CPU_LIST=1(or any others), all CPUs are used and when I delete the deconverlution layer, the environment variable works. Does the deconv_2d_blas.cpp cause this bug?

hey-yahei, 2019-07-04:
Hi, @Guanmoyu 
The executor with the implement of blas is not under control of Tengine, maybe you could set configuration for openblas library to bind your program with blas-op to cpu cores.
Thanks

Guanmoyu, 2019-07-14:
Thank you very much, I have tried this and run my example in the right CPU successfully!

closed 2019-08-08 by Guanmoyu