OAID/Tengine #155: cross compilation failed: cannot find -lprotobuf

PhilipXue, 2019-06-12:
Hi, I was cross-compling tengine inside a docker environment following the official note for cross compile, I have libprotobuf-dev installed on the host and compiled protobuf targeted at armv8 and copy all the files as instructed:
```
arm-cross-compile/tengine$ tree sysroot/ubuntu_rootfs/ -L 3
sysroot/ubuntu_rootfs/
`-- usr
    |-- include
    |   `-- google
    `-- lib
        `-- libprotobuf.so
```
But I still got this error:
```
/usr/lib/gcc-cross/aarch64-linux-gnu/7/../../../../aarch64-linux-gnu/bin/ld: cannot find -lprotobuf
```
I  am new to cross compilation. Can anyone help me figure out what was wrong? Thx

PhilipXue, 2019-06-13:
I figured out that in the Makefile, the ld difines as the following:
```
ifeq ($(CROSS_COMPILE),aarch64-linux-gnu-)
   SYSROOT_FLAGS:=--sysroot=$(SYSROOT) 
   SYSROOT_LDFLAGS:=-L/usr/lib/aarch64-linux-gnu -L/lib/aarch64-linux-gnu
   PKG_CONFIG_PATH:=$(SYSROOT)/usr/lib/aarch64-linux-gnu/pkgconfig
   export PKG_CONFIG_PATH
endif
```
so creating `aarch64-linux-gnu` foldf and move all the contents into this folder solves the problem.
closing this issue.

PhilipXue, 2019-06-13:
Sorry, I was wrong, the error still persists.
What should the `EMBEDDED_CROSS_ROOT` item be btw? I set this to be the root of toolchain that the manufacturer provides:
`/opt/FriendlyARM/toolchain/6.4-aarch64/`

PhilipXue, 2019-06-13:
I realized that I should not uncomment the `EMBEDDED_CROSS_ROOT` item. This makes the compile process without complaining about missing ld.
But this time when building tests, I encountered with `opencv2/imgproc/imgproc.hpp: No such file or directory`, but in `/usr/include/` folder all the head files exists:
```
ls /usr/include/opencv2/ | wc -l
78
```

hey-yahei, 2019-07-04:
Hi, @PhilipXue 
To build tests, opencv2 for arm should be provided at `Tengine_DIR/sysroot/ubuntu_rootfs/usr/lib/` and  `Tengine_DIR/sysroot/ubuntu_rootfs/usr/include/` just like protobuf.
Thanks