OAID/Tengine #158: Can't convert a easy TF model.

qimingyudaowenti, 2019-07-02:
Segmentation fault happen when I convert TensorFlow model (*.pb) to Tengine model:
```
[openailab@localhost tengine1.0_model_convert_tool_arm64]$ ./convert.sh -f tensorflow -m models/frozen_model2.pb -o models/a.tmfile
./convert.sh: line 5: 32402 Segmentation fault      (core dumped) ./convert_model_to_tm $*
```
To make sure the tool is ok, I test a pb model provided by Tengine: `mobilenet_v2_1.0_224_frozen.pb`, and get:
```
[openailab@localhost tengine1.0_model_convert_tool_arm64]$ ./convert.sh -f tensorflow -m models/mobilenet_v2_1.0_224_frozen.pb -o models/a.tmfile
Create tengine model file done: models/a.tmfile
```
My pb model is generated by the code:
```python
import tensorflow as tf
from tensorflow.python.framework import graph_util
import numpy as np

def net(input_image):
    conv1 = tf.layers.conv2d(input_image, 12, kernel_size=(3, 3), strides=1)  # (63, 63, 12)
    conv1_pool = tf.layers.max_pooling2d(conv1, (2, 2), 2)  # (30, 30, 12)
    conv2 = tf.layers.conv2d(conv1_pool, 12, kernel_size=(3, 3), strides=1)  # (28, 28, 12)
    conv2_pool = tf.layers.max_pooling2d(conv2, (2, 2), 2)  # (14, 14, 12)
    pool2_flat = tf.reshape(conv2_pool, [-1, 14 * 14 * 12])
    dense1 = tf.layers.dense(inputs=pool2_flat, units=1024, activation=tf.nn.relu)
    dense2 = tf.layers.dense(inputs=dense1, units=10)
    dense3 = tf.layers.dense(inputs=dense2, units=2, name='output')

    return dense3

input_image = tf.placeholder(tf.float16, shape=(None, 65, 65, 3))
output = net(input_image)

np.random.seed(7)

with tf.Session() as sess:
    sess.run(tf.global_variables_initializer())
    rand_array = np.random.randint(0, 255, size=(1, 65, 65, 3))
    print(sess.run(output, feed_dict={input_image: rand_array}))  # Will succeed.
    print(output.op.name)  # output/BiasAdd

    constant_graph = graph_util.convert_variables_to_constants(sess, sess.graph.as_graph_def(), ['output/BiasAdd'])

    with tf.gfile.GFile('tf_model/model_saved/method2/frozen_model2.pb', mode='wb') as f:
        f.write(constant_graph.SerializeToString())
```
The tensorflow version is 1.13. Can someone help me?

cyberfire, 2019-07-05:
@qimingyudaowenti 

please have a try if tengine can load and run this tensorflow pb directly?

Thanks,

Haitao

qimingyudaowenti, 2019-07-06:
@cyberfire
I use `tf_model.pb` and `tf_model.tmfile`(same file with different suffix). It gives same error info:
```
code: 0, mac addr is balabala
GetSN return 0
Wrong version of tm file
Create graph failed.
errno: 0
```
Does it help to fix?

cyberfire, 2019-07-11:
@qimingyudaowenti 

Can you share the .pb model with use?

Thanks,

Haitao

qimingyudaowenti, 2019-07-11:
@cyberfire 
Sure.

BaiDuPan：
  - link: https://pan.baidu.com/s/1ldzfXn1pjAf-e1oyJnCCwA 
  - password: 1qwe

GoogleDrive:
  - link: https://drive.google.com/drive/folders/1-lJhWoO76Nnvfpes16jSTnj2-Z6f2OCQ?usp=sharing

lyuchuny3, 2019-07-18:
Hi, @qimingyudaowenti
The current tengine model convert tool only support tensorflow_fp32 models.
You can change your codes by changing `tf.float16` to `tf.float32`:
```
    #input_image = tf.placeholder(tf.float16, shape=(None, 65, 65, 3))
    input_image = tf.placeholder(tf.float32, shape=(None, 65, 65, 3))
```
Then generate your pb file. Then you can convert it to tmfile.
If you want to convert the fp32-tmfile into fp16-tmfile,
you can run:
> build/tools/bin/quant_tm_model
[Usage]: ./quant_tm_model [-h] [-q quant_mode] [-i input_tmfile] [-o output_tmfile]
quant_mode: 0 -- fp16, 1 -- int8

closed 2019-07-19 by lyuchuny3