OAID/Tengine #161: Bus error

rahulsharma11, 2019-07-16:
Don't know why this error is with new Tengine-

Checking mobilenet_ssd and provided the necessary prototxt and caffemodel files in models.
But there is error-
ACL Graph Initialized
Driver: ACLGraph probed 1 devices
Bus error

This error is coming in "ret = run_graph(graph, 1);"
Any suggestion?

rahulsharma11, 2019-07-17:
Hi,
Bus error is now resolved. But there is another issue for Mobilenet_ssd_v2 caffemodel.
It is throwing error-
ACL Graph Initialized
Driver: ACLGraph probed 1 devices
cannot find load function for operator: Sigmoid
Create graph failed
 ,errno: 0

In previous Tengine release, i followed this link "https://github.com/OAID/Tengine/issues/89" and ended up with Segmentation fault.
I found in latest release sigmoid operator is included but still ther is error.
I am using rockpro64. ACL version - 18.05.
Any ideas?
Thanks

hey-yahei, 2019-07-19:
Hi, @rahulsharma11 
Something has been changed in latest version. Please follow #89 and register OP in `executor/operator/common/init.cpp` but not `executor/plugin/init.cpp` --    

```cpp
extern void RegisterSigmoid_NodeExec(void);
// ...
void RegisterCommonOps(void){
    // ...
    RegisterSigmoid_NodeExec();
}
```

If the segmentaion fault is still throwed, try to debug with **gdb**.

Thanks

Alicture, 2020-04-15:
@rahulsharma11 Have you solved your problem? Have a try with our latest version with ACL 19.02.

closed 2020-04-18 by BUG1989