OAID/Tengine #162: cannot load tensorflow model including BiLSTM OP

ForestLee, 2019-07-18:
[model_ner.zip](https://github.com/OAID/Tengine/files/3406884/model_ner.zip)
附件model_ner.zip是一个自然语言处理的命名实体识别的tensorflow模型文件，包含双向LSTM，可以用Netron打开看算子图。双向LSTM在自然语言处理领域是常用的网络，希望可以支持下，谢谢。后期我可以提供测试数据，做成example。

[model_ner_txt.zip](https://github.com/OAID/Tengine/files/3406915/model_ner_txt.zip)
附件model_ner_txt.zip是模型的text版本。

ForestLee, 2019-07-18:
模型很简单：输入层-->双向LSTM-->全连接层
我们实现的是命名实体识别，一般是由双向LSTM+CRF, 但是CRF层tensorflow没有对应算子，所以我们也没有包含进来，后期由用户app自己实现CRF。

lyuchuny3, 2019-07-19:
Hi, @ForestLee,
可以给出 生成这个简单的pb模型的python代码吗？

ForestLee, 2019-07-19:
我准备下示例代码。
现在还有如下小算子没实现：
cannot find load function for operator: ReverseSequence
cannot find load function for operator: Shape
cannot find load function for operator: RandomUniform
cannot find load function for operator: Floor
cannot find load function for operator: RealDiv
cannot find load function for operator: Pack

ForestLee, 2019-07-19:
[zh-NER-TF-master.zip](https://github.com/OAID/Tengine/files/3409784/zh-NER-TF-master.zip)
这个是在开源的基础上改了一点点。
解压后训练：python3 main.py train
就会产生模型文件。