OAID/Tengine #168: hit  issue when run prerun_graph, cannot find default executor

MicahXIE, 2019-08-12:
dev_allocator.cpp:296 cannot find default executor
generic_engine.cpp:100 dev executor allocator failed
graph_task.cpp:320 bad status: CREATED

I followed the example but I didn't set the device info like below. Could you help to give me some advice. Thanks

    /* set input shape */
    int node_idx = 0;
    int tensor_idx = 0;
    tensor_t input_tensor = get_graph_input_tensor(graph_, node_idx, tensor_idx);
    if(input_tensor == nullptr)
    {
        std::printf("Cannot find input tensor,node_idx: %d,tensor_idx: %d \n", node_idx, tensor_idx);
    }

    std::cout << "the image size is: " << nImg << std::endl;
    int dims[4] = {nImg, 3, height, width};
    set_tensor_shape(input_tensor, dims, 4);

    if(set_tensor_buffer(input_tensor, input_buffer.data(), input_size * sizeof(float)) < 0)
    {
        std::printf("Set buffer for tensor failed\n");
    }

    /* run the graph */
    prerun_graph(graph_);

    run_graph(graph_, 1);

    std::cout << "after run the graph !!!" << std::endl;

    // const char * output_tensor_name="InceptionV3/Predictions/Softmax";
    tensor_t output_tensor = get_graph_output_tensor(graph_, node_idx, tensor_idx);

    int dim_size = get_tensor_shape(output_tensor, dims, 4);
    if(dim_size < 0)
    {
        printf("get output tensor shape failed\n");
    }

hey-yahei, 2019-10-11:
Hi, @MicahXIE 
What model did you use in your application? Please check that if all operaters in it are supported by Tengine.

Thanks,
YaHei

liqi-c, 2020-04-14:
You can try again in new version . if didn't solve you problem , please mail to Support@openailab.com  or commit a new issue . thanks .

closed 2020-04-14 by liqi-c