OAID/Tengine #169: 关于多线程实现

fire717, 2019-08-13:
你好，我参考Tengine/tests/bin/mt_mssd.cpp.实现多线程，但是发现在run_graph这一步是阻塞的，即使多线程也会互相等待，这样并起不到加速的效果。

同时我尝试设置run_graph(graph, 0)改成非阻塞，然后通过wait_graph去获取结果，同样在wait_graph这一步也是要互相等待。

请问是我设置有问题还是有其他的多线程实现方式呢？谢谢。

MicahXIE, 2019-08-13:
你可以看看这个 https://github.com/OAID/Tengine/blob/master/doc/multi_thread_mssd.md

fire717, 2019-08-13:
> 你可以看看这个 https://github.com/OAID/Tengine/blob/master/doc/multi_thread_mssd.md

谢谢答复。

我看了，除了graph_t graph=create_runtime_graph这里和Tengine/tests/bin/mt_mssd.cpp中不一样，其他步骤都是一样的（都是先声明graph_t graph，然后set_graph_device，然后new std::thread()来执行）。

如果这里真的可以并行的话，那就说明是
`graph_t graph=create_runtime_grap `
和
`graph_t graph=create_graph`
这两个图声明方式的区别。

那么问题又来了，我要使用的是caffe的模型，需要传入prototxt和caffemodel两个路径，但是create_runtime_graph()的参数中只支持一个模型路径，应该是用于基于非caffe的框架的模型。

所以好像也解决不了我的问题。

另外，我是参考的官网http://www.tengine.org.cn/tutorial_display.php?id=20，这里的示例代码和你提到的那个markdown文档里是一样的。但是官网这里提到的源码就是源码在Tengine/tests/bin/mt_mssd.cpp.

MicahXIE, 2019-08-13:
detectMtcnn start!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
run_PNet start!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

run PNet set input tensor time: 0.005982
run PNet prerun graph time: 1.3e-05
run PNet run graph time: 1.57089
run PNet handle output tensor time: 0.001124

run PNet function time: 1.58254

你好 其实我也在做mtcnn 相关的部分, 发现每次处理时run_graph的时候也需要很长时间, 1.5s的延迟确实太高

fire717, 2019-08-13:
你这个是单线程吧？这个慢那就可能是因为本身就慢

MicahXIE, 2019-08-13:
嗯 是单线程 但是我是在PC端跑 和相同的设置的tensorflow和caffe模型比 慢很多 tf和caffe 大概200ms

fire717, 2019-08-13:
那是有点慢了 可能是tengine是专门针对移动端优化的吧

MicahXIE, 2019-08-13:
可是我在rk3399上跑 也挺慢的 不过我没开 ACL GPU 现在正在苦恼中...

liqi-c, 2020-04-14:
You can try again in new version . if didn't solve you problem , please mail to Support@openailab.com  or commit a new issue . thanks .
@MicahXIE @fire717

closed 2020-04-14 by liqi-c