OAID/Tengine #170: same results while running mxnet graph in example: "tengine/tests/bin/test_mxnet_mobileface.cpp"

devin730, 2019-08-20:
Hi,
I have built new tengine.so for android platform, make it capable to create graph for Mxnet. While runing "tengine/tests/bin/test_mxnet_mobileface.cpp" on RK3399 develop board, the results are unchanged, I have checked that the input date( which are got from different image ) are different.

Seems tengine is not capable to support mxnet??

I am using the mxnet model to get gender and age esimation, model is from github deepinsight/insightface, "insightface/gender-age/model/model-0000.params" and "model-symbol.json".

Thanks for assistance, looking for your reply.

lyuchuny3, 2019-08-21:
Hi, 
I have verified that using `test_mxnet_sqz.cpp`，Tengine can get two diff results with `cat.jpg` and `bike.jpg`.

For your problem,  my advice is:
* run **mxnet python code**, feeding two diff images for your  model, to see if the two results are the same.
* then compare the Tengine results with your python results.

devin730, 2019-08-27:
Hi, thank you for your advice, I have followed your advice. However, results are different while running mxnet python code, and Tengine results are still different. We have tried to transfer these mxnet model to caffe model or tengine model, still did not solve this problem.

It seems tengine.so on android platform has something I don't understand.

liqi-c, 2020-04-14:
@devin730 You can try again in new version . if didn't solve you problem , please mail to Support@openailab.com . thanks .

closed 2020-04-14 by liqi-c