OAID/Tengine #173: Makefile:198: recipe for target 'hclarm' failed

c936637238, 2019-09-22:
when I build the Tengine,I met a error, and I failed. How to fix it?
/home/cyf/tengine/scripts/makefile.build:119: recipe for target '/home/cyf/tengine/build/hclarm/../executor/operator/arm64/fc/sgemv_1x8_a72.o' failed
make[4]: *** [/home/cyf/tengine/build/hclarm/../executor/operator/arm64/fc/sgemv_1x8_a72.o] Error 1
make[4]: Leaving directory '/home/cyf/tengine/executor/operator/arm64/fc'
/home/cyf/tengine/scripts/makefile.build:71: recipe for target 'fc' failed
make[3]: *** [fc] Error 2
make[3]: Leaving directory '/home/cyf/tengine/executor/operator/arm64'
/home/cyf/tengine/scripts/makefile.build:71: recipe for target 'arm64' failed
make[2]: *** [arm64] Error 2
make[2]: Leaving directory '/home/cyf/tengine/executor/operator'
Makefile:86: recipe for target '../executor/operator' failed
make[1]: *** [../executor/operator] Error 2
make[1]: Leaving directory '/home/cyf/tengine/hclarm'
Makefile:198: recipe for target 'hclarm' failed
make: *** [hclarm] Error 2

hey-yahei, 2019-10-11:
Hi, @c936637238 
Which platform did you compile on? Maybe you are tring to compile arm64 version on x86 or arm32 platform.

Thanks,
YaHei

c936637238, 2019-10-12:
thank you ,finally I found that I compiled on my pc which is x64.

发自我的 iPad

c936637238, 2019-10-12:
I found I compiled x86version on x64,that’s why I failed.thank you for your help!

closed 2019-10-12 by hey-yahei