OAID/Tengine #174: how to run without protobuf?

cyf518, 2019-09-24:
How to transfer caffe model to tengine model, and compile it without protobuf?

hey-yahei, 2019-10-11:
Hi, @cyf518 
Protobuf is required by caffe_serializer and tf_serializer. To compile a Tengine without dependency of protobuf, you can comment `CONFIG_CAFFE_SERIALIZER=y`, `CONFIG_TF_SERIALIZER=y`, `CONFIG_TFLITE_SERIALIZER=y` and uncomment `CONFIG_TENGINE_SERIALIZER=y` in [makefile.config](https://github.com/OAID/Tengine/blob/master/makefile.config.example) file. Then convert caffe model to tengine model with the [convert tool](https://github.com/OAID/Tengine/blob/master/tools/bin/convert_model_to_tm.cpp) (you would find it in `<TENGINE_DIR>/install/tool/` directory) and load tengine model in your application.

Thanks,
YaHei

liqi-c, 2020-04-14:
You  can get convert tool from  release version : https://github.com/OAID/Tengine/releases . and then you didn't need compile without protobuf .

closed 2020-04-14 by liqi-c