OAID/Tengine #176: issue with onnx model test

rahulsharma11, 2019-10-15:
Hi, I am using latest Tengine on Rockpro64.
I wanted to test one onnx model that is converted from pytorch.
While testing "./build/tests/bin/test_onnx <model> <size>", it throws error-
Driver: ACLGraph probed 1 devices
infer shaped for node: 254 op: Convolution failed
graph_task.cpp:320 bad status: CREATED
graph_task.cpp:320 bad status: CREATED

Any suggestion what this error indicates? Is it because of model not converted properly ?
Thanks.

hey-yahei, 2019-10-15:
Hi, @rahulsharma11 
Have you converted model to onnx with official API of pytorch? Please double check if your onnx model is valid. And it would be better if you can provide the model file.

Thanks,
YaHei

rahulsharma11, 2019-10-15:
Hi, thanks for the reply,
I used the model from the followng link-
https://github.com/Daniil-Osokin/lightweight-human-pose-estimation.pytorch
(pretrained model - checkpoint_iter_370000.pth).

And i used official torch APIs for model conversion-

ex-
dummy_input = torch.randn(1, 3, 256, 256)
"torch.onnx.export(model, dummy_input, "model.onnx", verbose=True)"

rahulsharma11, 2019-10-15:
Note-
After conversion, there is one "Elu" operator in converted onnx model, was not default supported by Tengine. 
So i wrote the.

rahulsharma11, 2019-10-15:
Hi,
any update?
if you need i can provide my converted onnx model
Thanks.

hey-yahei, 2019-10-16:
Hi, @rahulsharma11 
Did you mean that you've implemented the Elu layer in Tengine by yourself? If so, maybe wrong output shape is infered by your Elu which cause the error in the subsequant Convolution layer. What's more, it is strange that the converted model(model.onnx) is only 16MB while the origin model(checkpoint_iter_370000.pth) 84MB, maybe you should check if the onnx model is correct.

Thanks,
YaHei

rahulsharma11, 2019-10-16:
Hi, 
**Did you mean that you've implemented the Elu layer in Tengine by yourself?**
Yes i doubted that also. 
So i converted all the "Elu" layer to "Relu" layer in converted model itself but got same error.

Furthermore i validated the converted onnx model with Netron visualizer. I did not find any corrupted layer. I would suggest if you can provide me the way for proper conversion from pytorch to onnx. 
Thanks.

hey-yahei, 2019-10-16:
Hi, @rahulsharma11 
I'm sorry that I'm not good at pytorch and onnx. But in general, the official API should be the best and most secure way. If pytorch support loading onnx model, it is suggested to load the converted model and then run it in order to check whether the model valid or not.

Thanks,
YaHei

rahulsharma11, 2019-10-16:
ok. 
Yes there are APIs to load the converted onnx model to validate and i did that same. Found no issue.
BTW will look for some more debug.

Thanks for the support.

closed 2019-10-16 by rahulsharma11