OAID/Tengine #179: cannot find load operator for ReLU6.

rahulsharma11, 2019-11-04:
Hi,
I can see there are operator implementation for ReLU6 in Tengine. But when i ran a caffemodel that uses ReLU6, it throws error-
cannot find load function for operator: ReLU6
create graph failed.

Any idea?
Thanks.

hey-yahei, 2019-11-06:
Hi, @rahulsharma11 
`ReLu6` [executor](https://github.com/OAID/Tengine/blob/master/executor/operator/common/relu6.cpp) is supported. But it seams that [caffe_serializer](https://github.com/OAID/Tengine/blob/master/serializer/caffe/caffe_serializer.cpp) doesn't parse `ReLU6` layer yet. Maybe you can add parse code to support ReLU6 layer for caffe model.

Thanks
YaHei

rahulsharma11, 2019-11-06:
Hi,
Thanks. I did that but after ReLU6, another operator support issue ocurred- Silence.

Meanwhile i was checking tflite examples also with downloaded tflite models from baidu webite.
I am running on rockpro64.
I tested all 3 examples (tf_lite_mssd, tf_lite_mssd_quant, tf_lite_mobilenet_quant). I think all of these should work with the downloaded models. But two examples throwimg error of "dev executor: arm.probed". Don't know what this error is.

I have also raised the issue -
https://github.com/OAID/Tengine/issues/180
Please have a look on above.

Is it still under development? or something i am missing.
Thanks.

rahulsharma11, 2019-11-13:
Hi,
i added LoadCaffeRelu6() parser to caffe_serializer.cpp according to the link you shared above.

But still getting error for 
Driver: ACLGraph probed 1 devices
cannot find load function for operator: ReLU6
Create graph failed
 ,errno: 0

Strangely this issue is with latest Tengine but its working with previous version of Tengine.

Alicture, 2020-04-15:
@rahulsharma11 I have checked our latest code, it's support ReLU6 in Caffe model. Pls try our latest version.

closed 2020-04-18 by BUG1989