OAID/Tengine #182: caffemodel prerun Core dumped

Lebron8997, 2019-11-08:
I used resnet18 net in caffe as follows. But it failed me when I run `build/tools/bin/convert_model_to_tm -f caffe -p SSD.prototxt -m SSD.caffemodel -o SSD.tm` 

Terminal:
`Create graph successfully!

prerun executor prepared!

sanitize finished.

set params finished.

get attr.

segmentation fault (core dumped)`

Debugging by GDB, I found that it failed at prerun. 

BTW, I tried the caffemodel using tengine directly with prerun error elsewhere. But how to fix it ?

Thank you!

[models.zip](https://github.com/OAID/Tengine/files/3822668/models.zip)

hey-yahei, 2019-11-08:
Hi, @Lebron8997 
Please check that if your model could be loaded and forwarded successfully in caffe.

Thanks,
YaHei

Lebron8997, 2019-11-08:
@hey-yahei I have checked it with opencv dnn and caffe. In fact, it was trained and converted from mxnet 1.5.0.

closed 2020-09-12 by BUG1989