OAID/Tengine #190: Library 'libcaffe-serializer.so' not found

yfang1644, 2019-11-29:
When I run ./build/test/bin/bench_sqz after build, it says:

Shared library not found: libcaffe-serializer.so: cannot open shared object file: No such file or directory
Get serializer failed, unknown model format: caffe
Create graph0 failed
errno: 2

Indeed, my system has no such file 'libcaffe-serializer.so'. So I think this library should be  built in
this project. And I modifiled default_config/...native.config, set 'BUILD_SERIALIZER' to 'y'.
Then I found these lines in linux_build.sh (lines  100-103):

cd tengine-module
mkdir build
cd build
../linux_build.sh

But directory 'tengine-module' is empty. the file '../linux_build.sh' doesn't exist!

erkaiyang, 2019-11-29:
@yfang1644 
hi，
Have you installed opencv? If not, you may get an error prompt when compiling the test file, please install opencv with the following command and try again：

`sudo apt install libopencv-dev  `

gunjupark, 2019-12-04:
maybe you were cloning "git clone https://github.com/OAID/tengine/"
instead that, you have to "git clone --recurse-submodules https://github.com/OAID/tengine/"

yfang1644, 2019-12-05:
Thank you @gunjupark. It does  work! And also thanks to @erkaiyang  for your quick reply.

linzeming, 2019-12-20:
@gunjupark  It's does not work after to "git clone --recurse-submodules https://github.com/OAID/tengine/",  Do you have other's way to solve this problem?

erkaiyang, 2019-12-21:
@linzeming 
Maybe you can try the following steps:

1. modify the file：`tests/bin/bench_sqz.cpp`
> //const char* model_file = "./models/squeezenet_v1.1.caffemodel";
> const char* model_file = "./models/squeezenet.tmfile";

> //graph_t graph = create_graph(nullptr, "caffe", text_file, model_file);
> graph_t graph = create_graph(nullptr, "tengine", model_file);
2. download [Tengine model](https://pan.baidu.com/s/1Ar9334MPeIV1eq4pM1eI-Q) (psw: hhgc)`squeezenet.tmfile` to the path`./Models/`
3. Recompile Tengine

xuguozhi, 2019-12-26:
`git clone --recurse-submodules https://github.com/OAID/tengine/`

`vim default_config/arm64_linux_native.config ` 

`BUILD_SERIALIZER=y`

`bash linux_build.sh default_config/arm64_linux_native.config`

Enjoy~

sunbinbin1991, 2020-01-09:
> @yfang1644
> hi，
> Have you installed opencv? If not, you may get an error prompt when compiling the test file, please install opencv with the following command and try again：
> 
> `sudo apt install libopencv-dev `

1: so if opencv not install, then libcaffe-serializer.so won't build? is that right?
2: so libmxnet_serializer.so not found is some reason?

closed 2020-04-15 by LEMON-LM