OAID/Tengine #191: Performance degradation after using GPU acceleration

xfan1024, 2019-12-04:
I run bench_mobilenet on RK3399 (Linux rk3399 4.4.179-ga179e620b389 #3 SMP Tue Dec 3 11:05:49 CST 2019 aarch64 GNU/Linux)

why cannot improving performance when I use `acl_opencl` device

no specify device
``` shell
[root@rk3399:/opt/tengine]# ./bin/bench_mobilenet 
run-time library version: 1.9.0
REPEAT COUNT= 100
Repeat [100] time 63681.25 us per RUN. used 6368125 us
8.6167 - "n02123159 tiger cat"
7.9423 - "n02119022 red fox, Vulpes vulpes"
7.8727 - "n02119789 kit fox, Vulpes macrotis"
7.3940 - "n02113023 Pembroke, Pembroke Welsh corgi"
6.3881 - "n02123045 tabby, tabby cat"
ALL TEST DONE
```

specify acl_opencl
``` shell
[root@rk3399:/opt/tengine]# ./bin/bench_mobilenet -d acl_opencl
run-time library version: 1.9.0
ACL Graph Initialized
REPEAT COUNT= 100
Repeat [100] time 122851.25 us per RUN. used 12285125 us
8.6167 - "n02123159 tiger cat"
7.9423 - "n02119022 red fox, Vulpes vulpes"
7.8727 - "n02119789 kit fox, Vulpes macrotis"
7.3940 - "n02113023 Pembroke, Pembroke Welsh corgi"
6.3881 - "n02123045 tabby, tabby cat"
ALL TEST DONE
```

rahulsharma11, 2020-01-06:
Hi,
Can you check with ACL version 19.02?
As i got decent performance for the same with acl_opencl.

closed 2020-04-22 by xfan1024

guoqiang0148666, 2020-05-02:
Me too @ Tengine: **v1.12.0**, ARM Compute Library: **v19.08**