OAID/Tengine #192: Tengine 1.9.0 Fully connected failed

gunjupark, 2019-12-04:
borad : Odroid N2 (arm64-v8a linux)

tengine_ver : 1.9.0

I use tengine/examples/imagenet_classification/classification.cpp -> Classify
(using caffemodel with libcaffe_serializer.so)

there is 2 major issue.

1. Fully connected layer Failed when prerun grpah stage.
```
ACL Graph Initialized
infer shaped for node: fc1000 op: FullyConnected failed 
Prerun graph failed
```
I tried 3 network that have FC layers (resnet50, alexnet, vgg16)
they all failed.

2. Classify's result is abnormal. (imagenet classificaion using Caffe model)
I ran 3 network
mobilenetv1, mobilenet v2, squeezenet v1.1 
their result all fault
[image]
[image]
[image]

How can I check this errors

I guess serializer's problem.

someone help me

gunjupark, 2019-12-04:
And these issue also appear in Note 10+ android adb shell inference

Alicture, 2020-04-15:
@gunjupark Please have a try with our latest code.If anything went south, let me know.

closed 2020-04-18 by BUG1989