OAID/Tengine #196: problem about convert model

QingyueMeng, 2019-12-21:
hi， I built the Tengin in Mac OS and Ubuntu 18.04 separately, and I test it in rk3399 use `adb` successfully , but I got the error like followings in both two platform when I try to convert the caffemodel to tmfile.
```
B000000119358A:convert mengqingyue$ ./convert_caffe_to_tm
-bash: ./convert_caffe_to_tm: cannot execute binary file
```

Could you give me same help?

erkaiyang, 2019-12-21:
@QingyueMeng 
hi，
You can download the model conversion tools [here](http://www.tengine.org.cn/info.php?class_id=105103).

QingyueMeng, 2019-12-23:
> @QingyueMeng
> hi，
> You can download the model conversion tools [here](http://www.tengine.org.cn/info.php?class_id=105103).

@erkaiyang hi, I download the tool from the link, and I run the command like this:
```
./convert_model_to_tm -h -f caffe -p mnasnet_192x144_1031_3.prototxt -m mnasnet_192x144_1031_3.caffemodel -o mnasnet_192x144_1031_3.tmfile
```

but I got the message like following:
```
[Usage]: ./convert_model_to_tm [-h] [-f file_format] [-p proto_file] [-m model_file] [-o output_tmfile]
```
 so, is there something wrong in the command? and can you give me an example about how to use it?

thank you

QingyueMeng, 2019-12-23:
and another question.
can I create my android studio project on Mac OS?

erkaiyang, 2019-12-23:
@QingyueMeng 
hi，

`-h` means help，try removing` -h`
 ```
./convert_model_to_tm -f caffe -p mnasnet_192x144_1031_3.prototxt -m mnasnet_192x144_1031_3.caffemodel -o mnasnet_192x144_1031_3.tmfile
 ```

QingyueMeng, 2019-12-23:
> @QingyueMeng
> hi，
> 
> `-h` means help，try removing` -h`
> 
> ```
> ./convert_model_to_tm -f caffe -p mnasnet_192x144_1031_3.prototxt -m mnasnet_192x144_1031_3.caffemodel -o mnasnet_192x144_1031_3.tmfile
> ```

@erkaiyang and there is another error.
```
tengine_model/convert$ ./convert_model_to_tm  -f caffe -p mnasnet_192x144_1031_3.prototxt -m mnasnet_192x144_1031_3.caffemodel -o mnasnet_192x144_1031_3.tmfile
The input file specified is using deprecated params: mnasnet_192x144_1031_3.prototxt
Please upgrade the input file by using caffe tools(upgrade_net_proto_text/upgrade_net_proto_binary).
Create graph failed
errno: 22
```

erkaiyang, 2019-12-23:
@QingyueMeng 

Your model is too old，and you can refer to FAQ11 in the link below
[https://github.com/OAID/Tengine/wiki/FAQ](https://github.com/OAID/Tengine/wiki/FAQ)

QingyueMeng, 2019-12-23:
> @QingyueMeng
> 
> Your model is too old，and you can refer to FAQ11 in the link below
> https://github.com/OAID/Tengine/wiki/FAQ

@erkaiyang I update my caffe model, but it throw the error above yet. 
And then I tried to use onnx, but it throw another error:
```
tengine_model/convert$ ./convert_model_to_tm -f onnx -m mnasnet_10313.onnx -o mnasnet_10313.tmfile
cannot find load function for operator: ConvTranspose
Segmentation fault (core dumped)
```

erkaiyang, 2019-12-24:
@QingyueMeng 
Tengine does not support this model. You can view the supported models [here](https://github.com/OAID/Tengine/wiki/%E6%94%AF%E6%8C%81%E7%9A%84OP).

closed 2020-05-10 by BUG1989