OAID/Tengine #197: Can't download from Model ZOO

mhsattarian, 2019-12-29:
Hi,
models in model zoo are hosted on Baidu which asks for a Chinese number to create an account, can you please provide another way to download them?

LEMON-LM, 2020-03-19:
ftp://ftp.openailab.net.cn/Tengine_models/

lutzroeder, 2020-04-09:
@LEMON-LM The `tmfile` folder is empty?

Is it possible to move the model zoo to a location like GitHub that can be accessed using `curl` and HTTP requests without passwords?

LEMON-LM, 2020-04-14:
> @LEMON-LM The `tmfile` folder is empty?
> 
> [image]
> 
> @BUG1989 Is it possible to move the model zoo to a location like GitHub that can be accessed using `curl` and HTTP requests without passwords?

You can get it from Google Drive now.
https://drive.google.com/open?id=1FE9Nz6EHHGZYZ1goYO5NcqiAWbLHRX0D

closed 2020-04-14 by LEMON-LM

pierricklee, 2020-04-15:
@LEMON-LM 
According to [Tengine/issue261](https://github.com/OAID/Tengine/issues/261), the tmfile format needs a fix, all model files have to be regenerated when we done. Please disable the access to Google Drive and reopen it after upload new model files. 

Cheers.

LEMON-LM, 2020-04-15:
> @LEMON-LM
> According to [Tengine/issue261](https://github.com/OAID/Tengine/issues/261), the tmfile format needs a fix, all model files have to be regenerated when we done. Please disable the access to Google Drive and reopen it after upload new model files.
> 
> Cheers.

ok,I`ll close it now.