OAID/Tengine #198: clEnqueueMapBuffer error with MSSD.

rahulsharma11, 2020-01-03:
Hi,
My platform is-
Rockpro64 with Ubuntu18.04
Tengine version 1.9.0 with ACL19.05

Code used- MT_MSSD

Issue-
1) While using 91 class COCO-mobilenetSSD model with ACL,
runtime error-
cpu_runner.cpp:801 graph: /home/rock64/rahul/mb_lite_coco_caffe_models/deploy_upgraded_withslice.prototxt:4 active num is: 3 output nodes: 2
cpu_runner.cpp:801 graph: /home/rock64/rahul/mb_lite_coco_caffe_models/deploy_upgraded_withslice.prototxt:3 active num is: 3 output nodes: 2
cpu_runner.cpp:801 graph: /home/rock64/rahul/mb_lite_coco_caffe_models/deploy_upgraded_withslice.prototxt:2 active num is: 3 output nodes: 2
cpu_runner.cpp:801 graph: /home/rock64/rahul/mb_lite_coco_caffe_models/deploy_upgraded_withslice.prototxt:1 active num is: 3 output nodes: 2
cpu_runner.cpp:801 graph: /home/rock64/rahul/mb_lite_coco_caffe_models/deploy_upgraded_withslice.prototxt:0 active num is: 3 output nodes: 2
terminate called after throwing an instance of 'cl::Error'
  what():  clEnqueueMapBuffer
Aborted

2) While using same 91 class COCO-mobilenetSSD model but without ACL,
it runs and give around 3 FPS speed.

Extra information-
1) The model has Relu6, multiple slice and sigmoid layer.
2) While using default 21 class mobilenetSSD model and following this thread-
https://github.com/OAID/Tengine/blob/master/doc/multi_thread_mssd.md
I am getting speed of around 16 FPS.

Any suggestions? Does ACL not support Relu6 or sigmoid? or something else i am missing.
Thanks.

rahulsharma11, 2020-01-07:
Hi,
update-
I found that it is not related to Relu or Relu6.
When environmetal variable "ACL_NHWC" is set, then only gpu is giving double speed with default 21-class Mobilenet_SSD.

But while using 91-class Mobilenet_SSD. It is throwing same error-
terminate called after throwing an instance of 'cl::Error'
what(): clEnqueueMapBuffer
Aborted

BUG1989, 2021-02-04:
We have updated the implement of ACL #540, please try is again, thanks.

closed 2021-02-04 by BUG1989