OAID/Tengine #203: a install problem about Openblas

knjwhn, 2020-03-04:
Hello , I've installed tengine from this guid on linux:
(https://github.com/OAID/Tengine/blob/master/doc/install.md)

I installed openblas from source code so  I edit the default_config/x86_linux_native.config :
OPENBLAS_LIB_PATH='-L/usr/local/OpenBLAS/lib -lopenblas'
OPENBLAS_INCLUDE_PATH='-I/usr/local/OpenBLAS/include'
and the compile process is successful

when I run ./build/benchmark/bin/bench_sqz , problem shows:
`symbol lookup error: /home/henan/tengine/build/libhclcpu.so: undefined symbol: cblas_sgemm`
 and I check the whether the libopenblas is in/usr/lib like:
```
/usr/lib$ ls -l | grep 'openblas' 
lrwxrwxrwx   1 root root      37 11月 21 12:51 libopenblas.a -> /usr/local/OpenBLAS/lib/libopenblas.a
lrwxrwxrwx   1 root root      38 11月 21 12:51 libopenblas.so -> /usr/local/OpenBLAS/lib/libopenblas.so
```
and I also make LD_LIBRARY_PATH and LIBRARY_PATH to /usr/local/OpenBLAS/lib/ and ldconfig.
However the problem still not solved.

and I close the option in x86_linux_native.config (OPEN_BLAS=n) it works.

and I found that the cmake warning show that:
```
CMake Warning:
  Manually-specified variables were not used by the project:

    ACL_ROOT
    AUTH_HEADER
    BUILD_SERIALIZER
    CONFIG_ACL_OPENCL
    CONFIG_TENGINE_ROOT
    ### OPENBLAS_LIB_PATH
    PROTOBUF_INCLUDE_PATH
    PROTOBUF_LIB_PATH
    USE_EXTERN_PATH
```

I was wondering wheather I didn't make the correct openblas PATH.

and I check the tengine_build.log,it shows ：
```
OPENBLAS_LIB_PATH=-L/usr/local/OpenBLAS/lib -lopenblas
OPENBLAS_INCLUDE_PATH=-I/usr/local/OpenBLAS/include
```
so I have nothing to do with it . Anyone's help would be appreciated.

Chambers1994, 2020-03-12:
And I have same Problem:
I hava complete build with these config

pynq@air-virtual-machine:~/Tengine$ vim default_config/x86_linux_native.config
OPENBLAS_LIB_PATH=/usr/include
OPENBLAS_INCLUDE_PATH=/usr/lib

However, meet this problem
pynq@air-virtual-machine:~/Tengine$ ./build/benchmark/bin/bench_sqz 
run-time library version: 1.12.0-github
./build/benchmark/bin/bench_sqz: symbol lookup error: /home/pynq/Tengine/build/libhclcpu.so: undefined symbol: cblas_sgemm

Can someone have help?

BUG1989, 2020-03-16:
The latest code is no longer depend on openblas, please try it again.

closed 2020-03-18 by BUG1989