OAID/Tengine #205: compatibility with opencv mat

soham24, 2020-03-07:


soham24, 2020-03-07:
I get buffer from camera feed, I read it as opencv mat and then rotate. 
I need to pass this in tengine Image format.
is there any compatibility?
I don't want to process extra copy operation.

LEMON-LM, 2020-03-20:
Why you need change opencv Mat to tengine ImageFormat?

LEMON-LM, 2020-03-20:
Why you need change Mat to tengine ImageFormat?
you can use the data of Mat to inference.
https://github.com/OAID/Tengine/blob/v1.3.2/examples/imagenet_classification/classification.cpp
you can refer to this example, how to use tengine with opencv.

BUG1989, 2020-04-04:
@soham24 
we have update the example with OpenCV Mat format
https://github.com/OAID/Tengine/blob/master/examples/classification_cv.cpp

closed 2020-04-04 by BUG1989