OAID/Tengine #213: can't convert Unsqueeze operator

EricSheng0402, 2020-03-14:
I try to convert pytorch to tm
first i convert pyotrch to onnx, because i have reshape in my pytorch model
my onnx file get operators like "unsqueeze, gather, shape"
which is not supported when i convert onnx to tm

BUG1989, 2020-03-16:
@EricSheng0402 Please use the [onnx-simplifier](https://github.com/daquexian/onnx-simplifier) to optimize the graph of onnx model, then try it again.

EricSheng0402, 2020-03-17:
@BUG1989 Thx for your replay. I tried this tool and succeed to convert the tm format model.

EricSheng0402, 2020-03-17:
@BUG1989 Thx for your replay. I tried this tool and succeed to convert the tm format model.
But when i use test_tm to verfy it, it didn't work.

sorry, i'm not good at C++, i try to debug it, the error happened in "LoadTmReshapeOp" function
and line 434: `for(unsigned int i = 0; i < v_re_shape->v_num; i++){`

Alicture, 2020-04-15:
@EricSheng0402  Convert your model to tmfile use the latest convert tool,and have a try with the latest version Tengine.

closed 2020-04-22 by BUG1989