OAID/Tengine #216: version 1.12 squeezenet with fully connected layer  Failed to execute on BatchNormalization1d

momandai, 2020-03-17:
system: x86 ubuntu18.04
tengine version: 1.12
create squeezenet with fc layer in pytorch, co

- [x]  create squeezenet with fc layer in pytorch
- [x]  convert pth to onnx,and simplify onnx
- [x]  convert simlified onnx to tm
- [ ]  run tm model with build/tests/bin/test_tm failed
`./tests/bin/test_tm /home/test/whale/repo/whcode/cv/whalemgs/onnx_tools/whale_reid_squeezenet_fc_sim.tm 256 128`
[image]

[whale_reid_squeezenet_model.zip](https://github.com/OAID/Tengine/files/4342099/whale_reid_squeezenet_model.zip)

momandai, 2020-03-17:
I think because  fc layer‘s batchnorm is 1d

BUG1989, 2020-03-18:
@momandai we have fixed the bug #220 , please try it again :)

closed 2020-03-19 by BUG1989