OAID/Tengine #221: [pytorch to onnx run in tengine error] ：Tengine-master/core/lib/graph_executor.cpp:357 infer shape failed on node: 454 due to input: 453 size is zero #49

fyq00, 2020-03-20:
Describe the bug
The bug is :
/home/ff/Tengine-master/core/lib/graph_executor.cpp:357 infer shape failed on node: 454 due to input: 453 size is zero

I trained my model with Pytorch and then transfered to onnx，using Ubuntu16.04.
When I use Tengine to run my model, bug appeared,like following:
These 3op are not supported
{shape, Gather, Unsqueeze}

Then I try onnx_simplifier,here is the new bug:
/home/ff/Tengine-master/core/lib/graph_executor.cpp:357 infer shape failed on node: 454 due to input: 453 size is zero

when I run API : dump_grap(grap);
It's following photo:
 
[image]

I don't know how to solve this problem~

BUG1989, 2020-03-21:
@fyq00  
We have fixed the bug in #226  , 
please try it again : )

closed 2020-03-21 by BUG1989

fyq00, 2020-03-22:
Thanks ! 
Now I can run my onnx model in Tengine.But the Model inference time becomes long！
Like the photo show：
I run the example of classification with mobilenetv2
[image]

It takes 5000ms!
and then I test mobienetssd

It takes 10000ms!

I change to the old version of Tengine and run the example of MobileNetSSD,
[image]
It only take 200ms~
My cpu seems too be fine~
Do you know whats problem about this?

#Ubuntu16.04 (with VMware)

fyq00, 2020-03-22:
Thanks ! 
But  I got some new problems, the inference time seems to longer than before~
https://github.com/OAID/Tengine/issues/221#issuecomment-602157187

 
Closed #221.

BUG1989, 2020-03-22:
I have fixed the bug of x86 ops not register, #227 : )

fyq00, 2020-03-22:
Thanks!
The bug is fixed!