OAID/Tengine #246: Compile completed in AIO RK-3399 C(AI), can't run the classification demo

LinkMple, 2020-04-01:
no matter compiled with aarch64-linux-gnu.toolchain.cmake or arm64.native.gcc.toolchain.cmake,  the classification demo still can't be executed.

mypath/Tengine/install/examples/ .classifacation -m xxx -i xxxx

[image]

i have compiled the  Tengine resource in x86 linux and succeed in runing /install/examples  classification demo, but why does this not work in my fire RK3399?  ε=(´ο｀*)))唉

liqi-c, 2020-04-02:
I'm sorry for hear that . 
Maybe you need compile tengine in Firefly again .  or you can compile in x86 by configure use aarch64-linux-gnu.toolchain.cmake .

LinkMple, 2020-04-02:
It seems to be worked after reboot  my RK3399.   The inference time in Mobilenet_v2 is average 80ms on 227*227 image, which faster than the same classification demo in x86-linux-cpu_only(i7 6700) platform.

Thanks for your suggestion. However the reason why the binary file can'nt run last time would not be discoverd. Maybe reboot and try again, for rookies, ACTUALLY solves almost problems.   /[Face with Tears of Joy]

liqi-c, 2020-04-03:
You can reappear the problem, and then find out the reason 
check the header information of the file through readelf -h or nm command  , and compare the differences .

closed 2020-04-14 by liqi-c

TUT-jiayou, 2021-04-07:
@LinkMple  Hi, 
which version of Ubuntu are you using in the rk3399.