OAID/Tengine #254: infer shaped for node: MobilenetV2/Conv/Conv2D-MobilenetV2/Conv/Relu6 op: Convolution failed

fathow666666, 2020-04-04:
First i use the mobilent model in Tengine model zoo,it is able to run.
Then I use the convert tool change tensorflow model to tengine model.   When use the mobilentv1.tmfile to classifay ,i says the error that   
infer shaped for node: MobilenetV1/MobilenetV1/Conv2d_0/Conv2D-MobilenetV1/MobilenetV1/Conv2d_0/BatchNorm/FusedBatchNorm-MobilenetV1/MobilenetV1/Conv2d_0/Relu6 op: Convolution failed
Prerun graph failed

[image]
[image]
[image]

liqi-c, 2020-04-07:
May you can try compare the differences between the two models？  through Netron tools .  
I will take a look at this problem .

LEMON-LM, 2020-04-07:
The input tensor shape of tensorflow model is [?,224,224,3],it means NHWC,
but in the test code ,the  definition  of input tensor shape is [1,3,224,224],  it means NCHW , you  need change the test code.And try it again.

liqi-c, 2020-04-07:
@fathow666666  the tensorflow model format is NHWC , but in our code default is nchw ,so you can modify like this : 

--- a/examples/classification.cpp
+++ b/examples/classification.cpp
@@ -67,7 +67,7 @@ bool run_tengine_library(const char* model_name, const char* tm_file, const char

     // set input
     int img_size = img_h * img_w * 3;
-    int dims[] = {1, 3, img_h, img_w};
+    int dims[] = {1, img_h, img_w ,3 };

closed 2020-04-14 by liqi-c