OAID/Tengine #261: Reshape offset_re_shape has incorrect value

lutzroeder, 2020-04-14:
Model zoo file `detect_tflite.tmfile` contains a `Reshape` operator at address `23244`.

`offset_t_param` points to this data at `23220`: `01 00 00 00 06 00 00 00 5B 00 00 00`.

`0x0000005B` is not a valid pointer for `offset_re_shape`?

@pierricklee @LEMON-LM @@BUG1989

pierricklee, 2020-04-14:
Hi, Lutz, 

Thanks for supporting Tengine!
Actaully, this tmfile model on google drive is out of date compare to the latest version of Tengine v1.12.
The data here you found is aligned by 
```
>     int32_t dim_0;
>     int32_t dim_1;
>     int32_t dim_2;
>     int32_t dim_3;
>     int32_t dim_size;
>     int32_t axis;`
```
[04b6791 ](https://github.com/OAID/Tengine/commit/04b679130770c059de74ae0e0bf6bdd998236cc6)has changed the TM2_ReshapeParam here:
[image]

The model on google drive used the previous format. So it makes sense with `0x0001 * 0x0006 * 0x005B`,  which equals `1 * 6 * 91`.

---

With the latest model: 
[image]

You could use this model: https://github.com/pierricklee/tmfile-sample/blob/master/detect_tflite.tmfile

@LEMON-LM Please update this model to google drive.

lutzroeder, 2020-04-14:
@pierricklee The old files still exist. Users will have it (the same happens for other files). They might use older versions of Tengine and might still open them in Netron which is now going to crash when the old format is loaded. 

This is not acceptable. If you change the format and want things to continue working you need to update version numbers (either for the operator or the model) so other tools can support both versions. If not, then tools like Netron can not support Tengine and we should remove support. Other tools can't be crashing because if there is no way to handle this.

pierricklee, 2020-04-15:
@lutzroeder That is a good suggestion! We are planning add an opset version and magic number into tmfile model. This will be on Tengine roadmap as soon as possible. I will keep you updated.

lutzroeder, 2020-04-15:
@pierricklee `op_ver` exists on `TM2_Operator` already (there is also model version stamp). It needs to be updated for breaking changes like this and all new files need to be regenerated now. The longer you wait, the more these incompatible identical looking files will spread causing problems.

pierricklee, 2020-04-15:
@lutzroeder That's the point. We are rewriting the `op_ver` part and will regenerate model files.

closed 2020-08-07 by lutzroeder