OAID/Tengine #264: problem with ResizeNearestNeighbor in tensorflow

wangg, 2020-04-15:
Tensorflow Version: 1.13.2
convert tool: compiled from the master branch of this repo

Observation:
It seems that the conversion of ResizeNearestNeighbor only works if it's followed by a conv layer.
Is this an expected behavior?

This model can be converted to tmfile sucessfully:

inputs = keras.Input(shape=(8, 8, 16), batch_size=1)
x = keras.layers.Conv2D(16, 3, strides=2, padding='same')(inputs)
x = keras.layers.UpSampling2D()(x)
x = keras.layers.Conv2D(16, 1, strides=1, padding='same')(x)
x = keras.layers.Add()([inputs, x])
model = keras.Model(inputs=inputs, outputs=x)

However, removing the conv layer after resizing will raise an error:

inputs = keras.Input(shape=(8, 8, 16), batch_size=1)
x = keras.layers.Conv2D(16, 3, strides=2, padding='same')(inputs)
x = keras.layers.UpSampling2D()(x)
x = keras.layers.Add()([inputs, x])
model = keras.Model(inputs=inputs, outputs=x)

error message:
error on load node: add/add op: Add
Create graph failed
errno: 0

Alicture, 2020-04-15:
@wangg Can you share your model with us? It's seems like a bug.

wangg, 2020-04-15:
This repo has everything to reproduce the problem.
model1.py, model2.py, model3.py generate the tensorflow models I mentioned above.
convert.sh shows how I call the Python scripts to generate tf models, then apply the convert tool.
Both model1 and model2 can be converted sucessfully, but model3 raises an error.

https://github.com/wangg/bug_report

wangg, 2020-04-15:
@Alicture 

I tried to run the converted model1 and model2.

Model 1 cant be converted to tmfile, but Tengine raises an error when running the model:

> Tengine/core/lib/graph_executor.cpp:357 infer shape failed on node: conv2d_1/Conv2D due to input: conv2d_1/bias size is zero

It seems that specified batch size + resize somehow caused a problem.

Model 2 can be run successfully.

wangg, 2020-04-16:
According to netron's visualization, when batch size is specified, the resize layer suddenly disappears after converted to tmfile.

Model1 (raises runtime error):
[image]

Model2 (runs as expected)
[image]