OAID/Tengine #267: DataLayout check fails when running NHWC model on acl_opencl

wangg, 2020-04-16:
A model with NHWC input cannot be run on acl_opencl because the data layout assertion fails.
When running on CPU only, everything works.

Tengine Version: master branch of this repo

Model: sequeezenet_tf.tmfile from Tengine model zoo
[squeezenet_tf.zip](https://github.com/OAID/Tengine/files/4484751/squeezenet_tf.zip)

Error message:
Tengine/tools/plugin/driver/acl_opencl/acl_graph.hpp:577: bool Tengine::CLGraph::AddConvolutionLayer(Tengine::Node*): Assertion 'TENGINE_LAYOUT_NCHW == TengineDataLayOut' failed when using

LEMON-LM, 2020-04-22:
I can`t reproduce this issue .
I had used the squeenet model of tensorflow,and the model of tengine,All of them can run on RK3399 with GPU .
Can you tell me your hardware and tengine version?

wangg, 2020-04-22:
I am also using RK3399, Tengine is compiled from the master branch of this repo.

I have attached the inference code and model I used. Could you please check if it runs as expected on your side? Please also kindly let me know if I made some newbie mistake in the code, which led to the problem.

[code_and_model.zip](https://github.com/OAID/Tengine/files/4515261/code_and_model.zip)

LEMON-LM, 2020-04-22:
You need use set_graph_device after prerun_graph.
Because in prerun_graph  we will set the layout for the model and all op.
You can see infershape fuction in each op,it will be call in prerun.

wangg, 2020-04-22:
If I use set_graph_device after prerun_graph, the model can run. However, it is running on CPU instead of GPU even though "ACL Graph Initialized" is printed out. (2 big cores of RK3399 are fully loaded and the GPU load stays at 0) The code I used is attached. Could you let me know if I did something wrong?

[set_device_after.zip](https://github.com/OAID/Tengine/files/4515916/set_device_after.zip)

ddzhao91, 2020-04-29:
@wangg Have you tried to set the environment “ACL_NHWC=1”
there is an environment variable “ACL_NHWC” used in file tools/plugin/driver/acl_opencl/acl_graph_device.cpp line 443

wangg, 2020-04-30:
@ddzhao91 Yes, I have tried it. If I call set_graph_device after prerun_graph, it is running on CPU no matter what the value of ACL_NHWC is. If I call set_graph_device before prerun_graph, the layout check fails. I also checked with one OAID  FAE through the QQ group, and NHWC models are actually not tested on GPU, so no guarantee for anything. Did you successfully run NHWC model on GPU without modifying Tengine? If so, I will really appreciate it if you can tell me how you made it.

BUG1989, 2021-02-04:
We have updated the implement of ACL #540, please try is again, thanks.

closed 2021-02-04 by BUG1989