OAID/Tengine #289: bench_sqz demo in arm cpu mode is faster than gpu mode (opencl)

guoqiang0148666, 2020-05-06:
Thanks for your excellent works.

l run bench_sqz demo in cpu mode, the inference speed is faster than gpu.

**CPU:**
```
root@linaro-alip:~/Tengine# ./build/benchmark/bin/bench_sqz -r 5
run-time library version: 1.12.0-github
REPEAT COUNT= 5
Repeat [5] time 63453.60 us per RUN. used 317268 us
0.2831 - "n02123045 tabby, tabby cat"
0.2714 - "n02123159 tiger cat"
0.1687 - "n02119789 kit fox, Vulpes macrotis"
0.0843 - "n02124075 Egyptian cat"
0.0750 - "n02085620 Chihuahua"
```
**GPU:**
```
root@linaro-alip:~/Tengine# ./build/benchmark/bin/bench_sqz -r 5 -d acl_opencl
run-time library version: 1.12.0-github
ACL Graph Initialized
REPEAT COUNT= 5
Repeat [5] time 144353.59 us per RUN. used 721768 us
1.0000 - "n04462240 toyshop"
0.0000 - "n01443537 goldfish, Carassius auratus"
0.0000 - "n01491361 tiger shark, Galeocerdo cuvieri"
0.0000 - "n01494475 hammerhead, hammerhead shark"
0.0000 - "n01440764 tench, Tinca tinca"
```

Is this normal phenomenon or some things l'm missing？Any suggestions will be welcome.

Thanks

BUG1989, 2020-05-10:
I see, the result of GPU is error, we will fix the problem as soon as possible.

BUG1989, 2021-02-04:
We have updated the implement of ACL #540, please try is again, thanks.

closed 2021-02-04 by BUG1989