OAID/Tengine #299: Does Tengine support deployment on windows?

shiwenloong, 2020-05-13:
I can't find CMAKE_TOOLCHAIN_FILE for windows in https://github.com/OAID/Tengine/tree/master/toolchains

liqi-c, 2020-05-18:
Windows system is not supported . You can try windows subsystem or  ubuntu virtual machine .
toolchains for linux system in PC  :  x86.gcc.toolchain.cmake and  x86_avx2.gcc.toolchain.cmake

kalcohol, 2020-09-03:
in addition, because of some low level code are based on gcc expand feature, one could build tengine or lite branch via MinGW or CygWin on windows(more like OpenBLAS windows version building process).
so strictly speaking, it's unhindered for building tengine on windows. but it's still bags of refactor work before that building tengine with MSVC is possible.

closed 2020-09-03 by kalcohol