OAID/Tengine #323: Performance on A53 Quad Core (S32V234)

kannanrn, 2020-06-17:
Hi

I checked the performance of standalone TEngine on A53 Quad Core (S32V234) / Linux for Yolo3-tiny (~34 MB weight file, input 416x416, image  is dog.jpg),  I observed inference time of 7000 ms.  While in OpenCV-DNN with the same Yolo3-tiny and input size, the inference time was 160 ms.  Also checked the perfromance with OpenCV-DNN (with Tengine), the inference time was 11000 ms.  Is something I am missing?

liqi-c, 2020-06-17:
We test yolov3-tiny with 416 * 416 input ,and 4*A53@1.4GHz ，inference time is 170ms .  
Mabye you compile error in tengine.  you can compile like this :  
cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/aarch64-linux-gnu.toolchain.cmake ..

kannanrn, 2020-06-17:
Thanks @liqi-c . I checked with the aarch64-linux-gnu.toolchain.cmake, looks the performance nos. are reasonable.  Earlier we used fsl toolchain to compile, seems compatible issue.   

Is it possible to quantize Yolo3-tiny.weight using the convert tool and use int8 model with TEngine.

kannanrn, 2020-06-18:
@liqi-c  Earlier I was checking with the custom Yolo3-tiny weight file, which is close to 6.8 MB, on 4*A53 @ 1GHZ it is taking close to 230 ms.  I suppose your performance no. of 160 ms is for the 34MB yolo3-tiny.wieghts. If so the performance difference is huge.

So I freshly tried converting the default 34 MB yolo3-tint.weight and cfg from below lin
https://pjreddie.com/media/files/yolov3-tiny.weights
https://github.com/pjreddie/darknet/blob/master/cfg/yolov3-tiny.cfg

Convertion tools failed - segmentation fault.  

While with the custom Yolo3-tiny weight (6.8 MB), I can do the conversion.  

Seeking your clarification on the below items
- Is there any size restriction with convertion tools?
- I searched for the yolo3-tiny.tmfile in ther repository, I couldn't find any thing, can you point out where is it?
- Can you please suggest how can I verify the performance issue?

Attaching the yolo3-tiny cpp test file, I had used for verification, let me know if this is ok?

[Yolo3_tengine.cpp.txt](https://github.com/OAID/Tengine/files/4798426/Yolo3_tengine.cpp.txt)

kannanrn, 2020-06-18:
[Yolo3_tengine.cpp.txt](https://github.com/OAID/Tengine/files/4798961/Yolo3_tengine.cpp.txt)
ignore the earlier sample test file.

liqi-c, 2020-06-19:
@kannanrn you can download from our model zoon : 
https://pan.baidu.com/s/1Ar9334MPeIV1eq4pM1eI-Q (psw: hhgc) 
Thanks .

liqi-c, 2020-06-19:
Is there any size restriction with convertion tools?  -- no 
Can you please suggest how can I verify the performance issue? 
  -- you test file  is ok .

kannanrn, 2020-06-19:
Sure @liqi-c. I will try the model from the link and update you.

Any reason why the yolo3 tiny conversion is failing? Even i tried the convert tool from the stable release that too gave segmentation fault.

kannanrn, 2020-06-19:
I did the testing with yolov3_tiny_dk.tmfile,  with the sample app(in the Yolo3_tengine.cpp.txt) i had created and mobilenetv2_yolov3_old_api in the examples.  

Results are varying
-sample app : 5331.67 ms 
-mobilenetv2_yolov3_old_api  : 301.78 ms

So to summarize, I see the below problems
- Convert tool is not working for default yolov3-tiny, I tried with different release, it simply wont work, segmentation fault. While it worked for one particular custom yolov3-tiny but not consistent.
- Performance difference between the 2 application ( my sample and example)

I am looking at the performance now so  did not check the output correctness.

----------------------------------------------------------------
Logs below
./Yolo3_tengine -m yolov3_tiny_dk.tmfile -i ../00dog.jpg

Inference Time : 5331.671875 ms
3.6876 - 186
3.6381 - 894
3.3668 - 952
3.3221 - 550
3.0802 - 213
--------------------------------------
ALL TEST DONE

./mobilenetv2_yolov3_old_api -m yolov3_tiny_dk.tmfile -i ../00dog.jpg
--------------------------------------

Repeat 1 times, avg time per run is 301.781 ms
max time is 301.781 ms, min time is 301.781 ms
pptime : 38.557999
num_detections,5
16: 57%
left = 110,right =  388,top = 158,bot =  544
7: 58%
left = 504,right =  648,top = 61,bot =  192
2: 65%
left = 419,right =  721,top = 69,bot =  177
1: 60%
left = 176,right =  605,top = 127,bot =  473
   done

kannanrn, 2020-06-19:
I did a fresh build for the convert file, yes i can do the yolo3-tiny conversion now.

liqi-c, 2020-06-19:
> I did the testing with yolov3_tiny_dk.tmfile, with the sample app(in the Yolo3_tengine.cpp.txt) i had created and mobilenetv2_yolov3_old_api in the examples.
> 
> Results are varying
> -sample app : 5331.67 ms
> -mobilenetv2_yolov3_old_api : 301.78 ms
> 
------------------------------------------------------------------------
C API is more low-level, and C ++ API is  just encapsulates a layer on C API.   
You use the same lib in two different api  ?

liqi-c, 2020-06-19:
Do not count the time of interface extract_tensor , compare the single layer with the run interface ？ 

    /* get result */
    detector.extract_tensor(0, 0, output_tensor);

kannanrn, 2020-06-19:
I use the same library to test both the C and C++ sample applciation.  

I did few more output testing with C application.

C application with low level API's is fast but the detection per frame dropped comparing the Darknet.   Tengine(17 objects)   vs  Darknet (28 Objects).   

Will the output match with the Darknet?  or it is getting reduced because of the optimization?  

How do I improve the detections?

liqi-c, 2020-06-28:
DUMP_NODE_OUTPUT(driver/cpu/cpu_runner.cpp)  macro can print out every layer output. 
You can use the same input, and then compare the output of each layer to find the different layer and find the reason,thanks  

If c and c++ application result is different ， you can attach the application file  , i will look this problem .

kannanrn, 2020-08-09:
> We test yolov3-tiny with 416 * 416 input ,and 4*[A53@1.4GHz](mailto:A53@1.4GHz) ，inference time is 170ms .
> Mabye you compile error in tengine. you can compile like this :
> cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/aarch64-linux-gnu.toolchain.cmake ..

I checked with the latest release, the performance for 416x416 on A52@1Ghz , inference time is ~925 ms (1 thread) / 646 ms(4 threads).   Earlier version was giving around ~301 ms (you can check the earlier threads).  I am not sure why there is a performance degradation with the latest version.

./tm_yolov3_tiny -m yolov3-tiny.tmfile -i dog.jpg
tengine-lite library version: 0.2-dev
Repeat 1 times, thread 1, avg time 925.44 ms, max_time 925.44 ms, min_time 925.44 ms
--------------------------------------
num_detections,2

./tm_yolov3_tiny -m yolov3-tiny.tmfile -i dog.jpg -t 4
tengine-lite library version: 0.2-dev
Repeat 1 times, thread 4, avg time 646.77 ms, max_time 646.77 ms, min_time 646.77 ms
--------------------------------------
num_detections,2

The inference timing of 160 ms on A53@1.5 ghz, is it open source or commercial license version?

closed 2021-01-06 by kannanrn