OAID/Tengine #327: 内存泄漏问题

enjustli, 2020-06-24:
https://github.com/OAID/Tengine/blob/v1.12.0/executor/operator/ref/ref_transpose.cpp#L91

长时间运行后，这个队列会不会非常大？

enjustli, 2020-06-24:
就是一个程序，不停的输入图片进行神经网络推理。连续跑几天几周这种

liqi-c, 2020-06-24:
op_param.permute.clear();
    for(int i = 0; i < tr_size; i++){
        op_param.permute.push_back(param_->tr_shape[i]);
    }
..........
    op_param.in_dims.clear();
    for(int i = 0; i < (int)in_dims.size(); i++){
        op_param.in_dims.push_back(in_dims[i]);
    }

麻烦加上如上的代码，然后测试看看，谢谢 ~  
如何可以，欢迎pr到我们master

skxie, 2020-06-25:
> op_param.permute.clear();
>     for(int i = 0; i < tr_size; i++){
>         op_param.permute.push_back(param_->tr_shape[i]);
>     }
> ..........
>     op_param.in_dims.clear();
>     for(int i = 0; i < [...]

这个`clear()`是不是最好加在`Postrun()`里面？

liqi-c, 2020-06-28:
> 这个`clear()`是不是最好加在`Postrun()`里面？

不用，在run的时候会多次调用，此时需要clear

enjustli, 2020-06-28:
添加这个后，已经没有泄露了

enjustli, 2020-06-28:
这样修改后，会对推理结果有影响么

liqi-c, 2020-06-29:
> 这样修改后，会对推理结果有影响么

不会的

closed 2020-07-03 by liqi-c