OAID/Tengine #328: convert_model_to_tm core dump

enjustli, 2020-06-24:
convert_model_to_tm 这个程序在x86 ubuntu上运行就崩溃，ldd convert_model_to_tm 这个程序，报错：not a dynamic executable

liqi-c, 2020-06-24:
##### linux x86_convert_tools
mkdir -p build-linux-x86
pushd build-linux-x86
cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/x86_convert_tool.gcc.toolchain.cmake ..
make -j$CPU_NUMS && make install
popd 

请采用如上的编译，x86设备上可以用gdb调试看看具体崩溃在哪 ？

codelilei, 2020-06-28:
@liqi-c
Segmentation fault **before entering main** function.
Here is the system information: 
Ubuntu 16.04.6 LTS
Linux 4.15.0-106-generic
GLIBC 2.27-3ubuntu1
gcc version 5.4.0 20160609
file convert_model_to_tm：ELF 64-bit LSB executable, x86-64, version 1 (GNU/Linux), statically linked, **for GNU/Linux 2.6.32**, BuildID[sha1]=445b3289f29d78aaec4f9262ca80f70c69e1b541, not stripped

It just works fine on another machine with the same configuration except that the glibc version is 2.23.

file convert_model_to_tm：ELF 64-bit LSB executable, x86-64, version 1 (GNU/Linux), statically linked, **for GNU/Linux 3.2.0**, BuildID[sha1]=44e7334c40d25fd3e7cd0eea6683d7d19291bff1, not stripped

icewaterq, 2020-06-30:
@liqi-c  你好，我在重新编译convert_model_to_tm的时候出现了错误，在执行cmake的时候报了下面的错误：
`-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CONFIG_ARCH_X86 = ON
-- CONFIG_ARCH_X86_AVX = OFF
-- CONFIG_ARCH_ARM64 = OFF
-- CONFIG_ARCH_ARM32 = OFF
-- CONFIG_TENGINE_SERIALIZER = ON
-- CONFIG_NCNN_SERIALIZER = OFF
GIT COMMIT ID: 0x655b93acd3ce02fd594b6173e996d0ea09833286
-- The ASM-ATT compiler identification is GNU
-- Found assembler: /usr/bin/as
-- CMAKE_TOOLCHAIN_FILE = /home/ice/code/tengine/toolchains/x86_convert_tool.gcc.toolchain.cmake
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Protobuf: /usr/local/lib/libprotobuf.so (Required is at least version "3.0.0") 
-- TENGINE_ROOT : /home/ice/code/tengine
CMake Error at tools/plugin/serializer/caffe/CMakeLists.txt:3 (message):
  caffe serializer needs PROTOBUF_LIB, please pass -DPROTOBUF_LIB=xxx when
  invoking cmake

-- Configuring incomplete, errors occurred!
`

在电脑里已经安装了protobuf，这里应该如何设置呢？

closed 2020-09-05 by BUG1989