OAID/Tengine #329: Is there an alternative way to download the models from Tengine Model Zoo?

ruck3rs, 2020-06-26:
Hello,
I am French and I am trying to download the models from Tengine model zoo. From my researches, it is not possible to download on pan.baidu.com from France.
Is there an other way for me to download the models? It would help me greatly!

Best regards.

liqi-c, 2020-06-28:
which models do you need .may we can upload it to github .

ruck3rs, 2020-06-28:
Thank you for your answer.
I am doing benchmark comparison between several approches to perform inference on embedded devices.
My goal is to test as many neural networks as possible. So I guess I need all the models from the Tengine Model Zoo.
If it is not possible, I would like to have at least squeezenet, resnet50, resnet18, mobilenet and mobilenetv2.
Would that be possible?

liqi-c, 2020-06-29:
Maybe you can use the benchmark  provide by Tengine.  when you finish compiling,  file ./build/benchmark/src/benchmark will be generated.  then you can run like this : 
squeezenet_v1.1 ：./build/benchmark/src/benchmark -s 1 
mobilenetv1： ./build/benchmark/src/benchmark -s 0 
resnet50： ./build/benchmark/src/benchmark -s 4 
mobilenetv2： ./build/benchmark/src/benchmark -s 7
...

ruck3rs, 2020-06-29:
Thank you, I have seen this but I would like to measure accuracy over different datasets as well which cannot be done with the not-pretrained benchmark models.

liqi-c, 2020-06-30:
Sorry, if you want to test the accuracy, you may need to test your own model. 
Currently, there is no other download address for model zoon . you can wait until the next version, and we will upgrade it ,thanks