OAID/Tengine #371: 赏金任务，第一期（快来薅羊毛~~）

BUG1989, 2020-08-21:
## **任务激励**

- 达成任意一项任务即成为 Tengine 开源贡献者、活动贡献者认证证书；
- 京东购物礼品卡（总共30个名额，先到先得）
  - 困难：1000 元
  - 一般：500 元
  - 简单：200 元

## **任务详情**

### 示例类

- 基于 Tengine  Lite 最新版本，实现 Yolov4 demo。（难度：一般）
- 基于 Tengine  Lite 最新版本，实现 OCR demo。（难度：一般）
- 基于 Tengine  Lite 最新版本，实现 MobileFaceNets demo。（难度：简单）
- 基于 Tengine  Lite 最新版本，实现 TensorFlow / TFLite 版本的 MobileNet-SSD demo。（难度：简单）

### API类

- 将原有 Tengine Python API 移植回 Tengine Lite，完成一个简单的分类示例即可。（难度：困难）
- 将原有 Tengine C++ API 移植回 Tengine Lite，完成一个简单的分类示例即可。（难度：一般）

### 优化类

- 基于 Tengine Lite 最新版本，优化 conv3x3s1 fp32 算子，实现 VGG16 网络模型单线程性能提升5%（例如： 1000 ms -> 950 ms）。（难度：困难）

### 文档类

- 基于 Tengine Lite 最新版本，在 examples/README.md 中补充 MobileNet-SSD、RetinaFace、Landmark、Yolact、Yolv3_Tiny 示例的运行方式及结果。（难度：简单）

BUG1989, 2020-08-24:
任务完成：将原有 Tengine Python API 移植回 Tengine Lite，完成一个简单的分类示例即可
#375

mengfu188, 2020-08-25:
`补充 MobileNet-SSD、RetinaFace、Landmark、Yolact、Yolv3_Tiny 示例的运行方式及结果。`

没看到这些模型文件在哪里啊，首页的百度网盘上面相关的ssd模型也不能直接运行，要用benchmark里面的模型文件吗？

shitouren1994, 2020-08-25:
github首页有：
模型仓库
Tengine model zoo 兼容原有 Tengine 的模型示例仓库（密码：hhgc）

BUG1989, 2020-08-25:
@mengfu188 我去过“七夕节”了，明天上午来更新~~不好意思。

BUG1989, 2020-08-27:
任务完成：基于 Tengine Lite 最新版本，实现 MobileFaceNets demo
#384

BUG1989, 2020-08-27:
@mengfu188 文档类-模型 已经更新至网盘链接，请查收，谢谢~

mengfu188, 2020-08-28:
@BUG1989 收到，谢谢

BUG1989, 2020-08-29:
任务完成：基于 Tengine Lite 最新版本，实现 Yolov4 demo
#389

BUG1989, 2020-08-29:
任务完成：
基于 Tengine Lite 最新版本，在 examples/README.md 中补充 MobileNet-SSD、RetinaFace、Landmark 示例运行方式及结果
基于 Tengine Lite 最新版本，在 examples/README.md 中补充 Yolact、Yolv3_Tiny 示例运行方式及结果
#386

BUG1989, 2020-08-29:
任务锁定，协助修改中：新增 mish activation op，完善 yolov4 demo
#390

BUG1989, 2020-08-30:
任务完成，基于 Tengine Lite 最新版本，实现 任意开源人体 Pose demo
#391

BUG1989, 2020-09-04:
任务完成，基于 Tengine Lite 最新版本，实现 x86 pooling 3x3/2x2 SSE 性能算子
#400

BUG1989, 2020-10-01:
任务完成，
基于 Tengine Lite 最新版本，优化 armv8.1 conv3x3s1 fp32 算子，VGG16 单线程性能提升5%
基于 Tengine Lite 最新版本，优化 armv7 conv3x3s1 fp32 算子，VGG16 单线程性能提升5%
#419

BUG1989, 2020-10-06:
任务完成， 基于 Tengine Lite 最新版本，实现 OCR demo
#429

chenjun2hao, 2020-10-13:
请问还有第二期吗？第一期没有赶上。期待中。

BUG1989, 2020-10-13:
@chenjun2hao 第二期任务，将在10月19日，请随时关注呀

BUG1989, 2020-12-21:
任务完成，基于 Tengine Lite 最新版本，实现在 Windows10 + Visual Stdio 编译功能 #508