OAID/Tengine #372: Segmentation fault

hufangjian, 2020-08-21:
你好，我把tengine转到androi环境上报 Segmentation fault 错误

cpu:64-bit ARM Cortex-A53
ndk:android-ndk-r18b
编译命令：cmake -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build-android-aarch64/cmake/android.toolchain.cmake -DANDROID_ABI="arm64-v8a" -DANDROID_PLATFORM=android-21 ..
make

BUG1989, 2020-08-21:
您好，请问是使用的 Tengine Lite 吗？如果不是，请使用最新版本的 Tengine Lite 再尝试，谢谢。

hufangjian, 2020-08-21:
您好，我是用最新版的Tengine Lite 

这是我的安卓版本信息。
[image]

ndk是android-ndk-r18b
编译 cmake -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build-android-aarch64/cmake/android.toolchain.cmake -DANDROID_ABI="arm64-v8a" -DANDROID_PLATFORM=android-21 ..

运行：

adb push ./lib/libtengine-lite.so /data/local/tmp/
adb push ./bin/tm_classification /data/local/tmp/
adb push ./bin/cat.jpg /data/local/tmp/
adb push ./bin/squeezenet_caffe.tmfile /data/local/tmp/

adb shell
cd /data/local/tmp/
export LD_LIBRARY_PATH=.
./tm_classification -m squeezenet_caffe.tmfile -i cat.jpg

结果：

msm8953_64:/data/local/tmp $ ./tm_classification -m squeezenet_caffe.tmfile -i cat.jpg -l synset_words.txt
Image height not specified, use default 227
Image width not specified, use default  227
Scale value not specified, use default  1.0, 1.0, 1.0
Mean value not specified, use default   104.0, 116.7, 122.7
tengine-lite library version: 0.2-dev
Segmentation fault

是否是我的squeezenet_caffe.tmfile 模型不正确？
我在ubuntu18.04 系统上编译是可以运行的。谢谢！

hufangjian, 2020-08-21:
是否是因为openMP的原因？我在X86系统上装了openMP 安卓里面没有。
[image]

hufangjian, 2020-08-24:
> @hufangjian 我们已经修复好了，麻烦您再试试？
已经严重可以了。非常感谢。