OAID/Tengine #379: 增加自定义算子的疑问

zjd1988, 2020-08-24:
上周末尝试增加yolov4中的mish运算符，需要同时在https://github.com/OAID/Tengine-Convert-Tools和https://github.com/OAID/Tengine两处进行增加，感觉不太方便，后续会有合并转换和推理代码的计划吗？
还有就是新增自定义算子，修改文件较多，而且并未找到一个示例增加代码，只有步骤（对于新手来说，有代码往往会更容易上手，也许是我没有找到^_^），不过还是希望tengine越来越好，后续会继续关注。

bzhang5, 2020-08-24:
谢谢你的建议，对于此问题后续我们会想方法改进

bzhang5, 2020-08-24:
@zjd1988  谢谢你的提议，对于此问题后续我们会想方法改进，也会在我们的开发计划里面

BUG1989, 2020-08-24:
@zjd1988 您好，请问能提供一个联系方式吗？我们这边协助您完成 PR，谢谢

zchrissirhcz, 2020-08-24:
尝试新增算子时，发现结构体定义有冗余，例如`DetectionOutputParam`和`TM2_DetectionOutputParam`，是不是可以合并起来。或者能否解释一下，分开定义的好处有哪些。

`/home/zz/work/Tengine-Convert-Tools/operator/include/operator/detection_output_param.hpp`:
```c++
struct DetectionOutputParam : public NamedParam
{
    int num_classes;
    int keep_top_k;
    int nms_top_k;
    float confidence_threshold;
    float nms_threshold;

    DECLARE_PARSER_STRUCTURE(DetectionOutputParam)
    {
        DECLARE_PARSER_ENTRY(num_classes);
    };
};
```

`/home/zz/work/Tengine-Convert-Tools/serializer/include/tengine/v2/tm2_format.h`:
```c++
typedef struct
{
    int32_t num_classes;
    int32_t keep_top_k;
    int32_t nms_top_k;
    float confidence_threshold;
    float nms_threshold;
} TM2_DetectionOutputParam;
```

zjd1988, 2020-08-25:
@BUG1989 email ：942002795@qq.com 手机：13816001614

BUG1989, 2020-08-26:
> 尝试新增算子时，发现结构体定义有冗余，例如`DetectionOutputParam`和`TM2_DetectionOutputParam`，是不是可以合并起来。或者能否解释一下，分开定义的好处有哪些。
>
> `/home/zz/work/Tengine-Convert-Tools/operator/include/operator/detection_output_param.hpp`: [...]

我认为这是冗余设计的，但是这个 Tengine Convert Tools 是源之于之前的 Tengine 项目，后续我们会对冗余设计进行重构，谢谢您的反馈。

BUG1989, 2020-09-01:
#390

closed 2020-09-01 by BUG1989