OAID/Tengine #381: tengine-lite   android 多核计算速度没有提升

hufangjian, 2020-08-24:
您好，我的android编译命令：
cmake -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake -DANDROID_ABI="arm64-v8a" -DANDROID_PLATFORM=android-22 -DANDROID_STL=c++_shared -DANDROID_ARM_NEON=ON -DCONFIG_ARCH_ARM64=ON -DANDROID_ALLOW_UNDEFINED_SYMBOLS=TRUE ..

在android设备上测试的时候：
[image]
[image]
[image]
几次测试CPU使用率几乎都是一样的。
[image]

之前测试的旧版本的Tengine，多核的清空下速度会提升很多。最新的tengine-lite速度基本上没提升。

BUG1989, 2020-08-26:
请问是 哪个模型多核计算速度异常？android32 还是 android64 呢？谢谢~

hufangjian, 2020-08-31:
您好，问题已经解决是我编译问题。谢谢。

closed 2020-09-01 by BUG1989